rt-net/crane_x7_ros #84: melodic対応

nomumuRT, 2019-04-23:
CRANE-X7のROSパッケージはどうやらmelodicでも動くようだが、社内でエビデンスが取れてない。
正式にサポートを表明したいので、動作検証してkineticと異なる箇所があればbranchを分けて対応コードを公開したいと考えています。

ShotaAk, 2019-05-07:
実行時に

> xacro: in-order processing became default in ROS Melodic. You can drop the option.
と表示されるが、エラーでも警告でもないので問題なし。

ShotaAk, 2019-05-08:
実行直後と、rvizの起動後に

> Kinematics solver doesn't support #attempts anymore, but only a timeout.

と警告が表示される。
動作はするので致命的な問題ではない。

https://github.com/ros-planning/moveit/pull/1288 により、attemptsオプションが削除された。

crane_x7_moveit_config/config/kinematics.yaml b/crane_x7_moveit_config/config/kinematics.yamlを下記のように修正すれば、実行直後の表示は消える。

```diff
   kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin
   kinematics_solver_search_resolution: 0.005
   kinematics_solver_timeout: 0.005
-  kinematics_solver_attempts: 3
```

rviz起動後の表示は残る。こちらは、moveitのrviz pluginがattemptsオプションを使用しているため表示される。

ShotaAk, 2019-05-08:
サンプルがすべて動作することを確認した。

ShotaAk, 2019-05-09:
gazebo起動時にエラーが発生。

> [ERROR] [1557375849.440592228, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_shoulder_fixed_part_pan_joint
[ERROR] [1557375849.441365356, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_shoulder_revolute_part_tilt_joint
[ERROR] [1557375849.442059141, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_upper_arm_revolute_part_twist_joint
[ERROR] [1557375849.442700730, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_upper_arm_revolute_part_rotate_joint
[ERROR] [1557375849.443410635, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_lower_arm_fixed_part_joint
[ERROR] [1557375849.444097991, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_lower_arm_revolute_part_joint
[ERROR] [1557375849.444696407, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_wrist_joint
[ERROR] [1557375849.445253096, 0.333000000]: No p gain specified for pid.  Namespace: /crane_x7/gazebo_ros_control/pid_gains/crane_x7_gripper_finger_a_joint

ShotaAk, 2019-05-10:
Inertiaの表示がおかしい。
パラメータの表記方法が変更されたかもしれないので、調査する。
[image]

commit 59ec7071 (2019-08-27): Merge pull request #93 from rt-net/melodic

support ROS Melodic. Close #84
closed 2019-08-27 by ShotaAk