rt-net/crane_x7_ros #107: Why are there two types of joints?

MapleHan, 2019-12-21:
Hello,
Gripper joint is Position Joint, why use Effort Joint?("arg use_effort_gripper")
Thankyou!

ShotaAk, 2019-12-23:
Hello,

`use_effort_gripper` is used for [*pick_and_place_in_gazebo_example*](https://github.com/rt-net/crane_x7_ros/blob/master/crane_x7_examples/README.en.md#pick_and_place_in_gazebo_examplepy).

If the argument is true, the gripper hardware interface will be `EffortJointInterface`.
https://github.com/rt-net/crane_x7_ros/blob/6275b0e7c5fa8e52be8aadd0b57227edf483604d/crane_x7_description/urdf/crane_x7_wide_two_finger_gripper.xacro#L51-L58

MapleHan, 2019-12-23:
Use EffortJointInterface for "pick_and_place_in_gazebo_example",Why not PositionJointInterface?

ShotaAk, 2019-12-23:
This is to enable `max_effort` option.
https://github.com/rt-net/crane_x7_ros/blob/6275b0e7c5fa8e52be8aadd0b57227edf483604d/crane_x7_examples/scripts/pick_and_place_in_gazebo_example.py#L53

This example can pick up the small and large object without change the gripper angle value.
https://github.com/rt-net/crane_x7_ros/blob/6275b0e7c5fa8e52be8aadd0b57227edf483604d/crane_x7_examples/scripts/pick_and_place_in_gazebo_example.py#L34-L35

If X7 is picking up a object in small gripper angle (e.g. GRIPPER_CLOSE = 0.1), the effort value of the gripper will reach a limit value (`max_effort`) then X7 stops closing the gripper angle and pick up the object successfully.

Please try to change the `max_effort` and `GRIPPER_CLOSE` values.

MapleHan, 2019-12-23:
ok,I understand,**thankyou**!!!

closed 2019-12-26 by MapleHan