rt-net/crane_x7_ros #118: Dynamixelのoperating modeを電流制御モードに変更できない

mayataka, 2020-07-07:
現在，こちらのリポジトリの内容を参考にさせていただき全関節を電流制御しようとしています．
しかし，crane_x7_control/config/crane_x7_control.yamlをハンドの設定を参考に，すべての関節においてoperating mode: 0にしても，実機のDynamixelは位置制御モードとして立ち上がってしまいます．
また，その際の各Dynamixelの設定をrqtで設定を確認すると，全ての関節の設定がoperating mode: 3となっています．

実行環境：Ubuntu 18.04，ROS melodic

まず，このリポジトリの仕様上，ハンド以外は位置制御に限定されているのかどうかをお聞きしたく思っております．
そうでない場合は，何かしたこのエラーが発生するヒント等を教えていただけると幸いです．

ShotaAk, 2020-07-08:
Issueの作成ありがとうございます。

> しかし，crane_x7_control/config/crane_x7_control.yamlをハンドの設定を参考に，すべての関節においてoperating mode: 0にしても，実機のDynamixelは位置制御モードとして立ち上がってしまいます．
> また，その際の各Dynamixelの設定をrqtで設定を確認すると，全ての関節の設定がoperating mode: 3となっています．

ROSパッケージ側の手順は間違ってないように見えます。

サーボモータ自体の制御モードは電流モードになっているでしょうか？
[Dynamixel Wizard 2](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) 等のアプリケーションで確認お願いします。

参考：
https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11

mayataka, 2020-07-08:
ご回答ありがとうございます．
ひとまず本issueはcloseさせていただきます．

closed 2020-07-08 by mayataka