rt-net/crane_x7_ros #154: コントリビュータの皆様へ：ライセンス変更とcrane_x7_descriptionの移動について

ShotaAk, 2022-05-12:
本リポジトリには[非商用ライセンス](https://github.com/rt-net/crane_x7_ros/blob/429d3292b79f5a2b787b4c429e9f00d71aefffc9/LICENSE)が適用されています。
パッケージの使いやすさを考慮して、**crane_x7_description以外の**パッケージに[Apache-2.0](https://www.apache.org/licenses/LICENSE-2.0)ライセンスを適用します。

具体的には次のようにパッケージ構成を変更します。

## 変更前

- crane_x7_rosリポジトリの**すべてのパッケージ※に非商用ライセンス**が適用される
  - [タグ：v1.0.0](https://github.com/rt-net/crane_x7_ros/releases/tag/v1.0.0)としてリリース済み

※具体的には以下のROSパッケージです
- crane_x7（メタパッケージ）
- crane_x7_bringup
- crane_x7_control
- crane_x7_description
- crane_x7_examples
- crane_x7_gazebo
- crane_x7_moveit_config
- crane_x7_msgs

## 変更後

- **crane_x7_descriptionパッケージを新規リポジトリへ移動する**
  - ライセンスには、引き続き**非商用ライセンス**を採用する
- crane_x7_rosリポジトリからcrane_x7_descriptionパッケージを削除し、**Apache-2.0**ライセンスを適用する
  - **タグ：v2.0.0**としてリリースする

## 注意事項

- crane_x7_descriptionの移動によって、パッケージのビルド手順や更新手順が変更されます。
  - 具体的な作業手順については、このissueのコメントとREADMEに記載します。
- crane_x7_descriptionには**引き続き非商用ライセンスが適用**されます。
- v1.0.0以前のソフトウェアのライセンスが**変更されることはありません。**
    - Apache-2.0ライセンスでソフトウェアを使いたい場合はv2.0.0以降をご利用ください。

---

@gbiggs  パッケージ作成時にコントリビュートして頂いたため連絡します。特に意見がなければ :+1:  お願いします。

gbiggs, 2022-05-15:
賛成です。

Tiryoh, 2022-05-26:
# TODO

- [crane_x7_descriptionリポジトリ](https://github.com/rt-net/crane_x7_description)の公開
    - [x] https://github.com/rt-net/crane_x7_description/pull/1
    - [x] https://github.com/rt-net/crane_x7_description/pull/2
- crane_x7_descriptionをcrane_x7_rosリポジトリから削除
    - [x] https://github.com/rt-net/crane_x7_ros/pull/155
- [ ] crane_x7_rosのライセンス変更

ShotaAk, 2022-05-27:
[v2.0.0](https://github.com/rt-net/crane_x7_ros/releases/tag/v2.0.0) をリリースしました。

パッケージの更新方法については[README](https://github.com/rt-net/crane_x7_ros/blob/v2.0.0/README.md) を参照してください。

closed 2022-09-06 by ShotaAk