rt-net/crane_x7_ros #202: Humble版でのビルドについて

s-kajita, 2025-01-19:
# 現状の問題点
Ubuntu 22.04LTS, ROS2 Humbleの環境で、crane_x7_rosのインストールで問題に遭遇しました。

現状のInstallation/Build from source に従うとビルドに失敗し、以下のエラーが出ます。
[image]

# 改善案
git cloneの際のブランチをros2 から humble-devel に変更した結果、ビルドできました。

（現状）
$ git clone -b ros2 https://github.com/rt-net/crane_x7_ros.git

（改善案）
$ git clone -b humble-devel https://github.com/rt-net/crane_x7_ros.git

Kuwamai, 2025-01-20:
ブランチ名について補足させていただきます。
最近のROSパッケージでは`-devel`を含まないディストリビューション名のみのブランチ名が主流となっているため、今回のJazzy対応に伴いHumble版においても`humble`ブランチを作成しました。今後のバグ修正等は`humble-devel`ブランチでなく`humble`ブランチで対応いたしますので、そちらも合わせてご確認お願いします。
https://github.com/rt-net/crane_x7_ros/tree/humble

s-kajita, 2025-01-21:
もう一点。

Humble環境でインストールすると RealSenseのトピック名が
/camera/camera/color/image_row
/camera/camera/color/camera_info
となります(/cameraが２重になる)。

このためデモを動作させるために、rvizの設定と aruco_detection.cpp のトピック名の
修正が必要でした。

Kuwamai, 2025-01-21:
@s-kajita 
ありがとうございます。
ros2ブランチにて修正した不具合が`humble-devel`ブランチに反映されておりませんでした。
お手数ですが`humble`ブランチに切り替えたうえでご確認いただけないでしょうか。

s-kajita, 2025-01-21:
crane_x7_ros を humble-develから humble ブランチに切り替えてテストし、問題なく動作することを確認しました。

有難うございました！

s-kajita, 2025-01-21:
アールティBlogでは、現在 humble-devel がリンクされているので、ここを修正されると私のようにはまる人が出なくなると思います。
https://rt-net.jp/humanoid/archives/5069

[image]

Kuwamai, 2025-01-21:
ありがとうございます。
修正させていただきました。
ドキュメントの更新が間に合っていない部分もあるため、順次更新させていただきます。

s-kajita, 2025-01-21:
確認しました。有難うございました！

closed 2025-01-21 by s-kajita