rt-net/crane_x7_ros #204: サーボモータ可動範囲を変更したい

s-kajita, 2025-01-23:
グリッパの原点がずれているようで、０度を設定しても完全にグリッパが閉じない不具合があります。
実機に合わせて可動範囲を修正するために crane_x7_ros/crane_x7_control/config/manipulator_links.csvの
サーボモータ可動範囲θ1,θ2の値を変更しましたが、反映されないようです。

環境：Ubuntu 22.04, ROS2
ブランチ： humble
実行コマンド： ros2 launch crane_x7_examples demo.launch.py port_name:=/dev/ttyUSB0 use_d435:=true

（参考）
demo.launch.pyの53行目から以下の記述があるので、manipulator_links.csvは読み込まれている気が
するのですが・・・

    links_file_path = os.path.join(
        get_package_share_directory('crane_x7_control'),
        'config',
        'manipulator_links.csv'
    )

Kuwamai, 2025-01-23:
お問い合わせありがとうございます。
サーボの原点位置を調整する場合は、DYNAMIXEL Wizardを使用してHoming offsetを調整お願いします。

DYNAMIXEL Wizard
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/

Homing offsetについて
https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#homing-offset

s-kajita, 2025-01-23:
有難うございます。
原点位置修正は、Homing offsetの設定で対処すべき件、了解いたしました。

別件ですが、manipulator_links.csv によってPIDゲインの修正が可能に見えますが正しいでしょうか？
肘関節(Link4)の位置決めエラーが大きいため、Pゲインを増やしたいと思っています。

Kuwamai, 2025-01-23:
ROS 2パッケージから`manipulator_links.csv`にて定義されているゲイン設定は使われておらず、`crane_x7_hardware.cpp`にてパラメータ設定をしております。各軸ごとに別々のパラメータを設定する場合は`crane_x7_hardware.cpp`を編集する必要があります。
https://github.com/rt-net/crane_x7_ros/blob/c565ff6446b014a8d9dfaa51268cf7ae44085b93/crane_x7_control/src/crane_x7_hardware.cpp#L28-L34

s-kajita, 2025-01-23:
了解です。有難うございました！

closed 2025-01-27 by Kuwamai